bertmelis/esp32ModbusTCP #4: onData/onError callbacks are not called in the example code.

thanakritlee, 2019-05-27:
Hello Bert,

I'm trying to run the example code given with my device/WiFi setup. I'm able to connect to the WiFi, and when readHoldingRegisters() is called I get a positive packetId back (as expected). However, the onError() and onData() callbacks are never called.

Do you have any idea what the problem might be?

Regards,
Thanakrit Lee

bertmelis, 2019-05-29:
could you try with logging enabled?

HilverinkJ, 2019-08-16:
Hello Bert,
I'm trying to read data from 3 eastron sdm220 energy meters with slave ID 1,2,3 with your sample code. Wifi is connected. Al the displayed values are zero.
It's not clear for me if this must be possible. 

Rgards,
Joop