bertmelis/esp32ModbusTCP #6: expected declaration before '}' token

zekageri, 2019-08-14:
In first compile with the librari : 
expected declaration before '}' token in the esp32ModbusTypeDefs.h line 66.

bertmelis, 2019-08-14:
I can't see where this error comes from...
How does your file compare to the file in this repo?
Are there any more error messages?

Remark: I had some problems with this lib and am rewriting completely from scratch.

zekageri, 2019-08-14:
Iam using the Example code from github. I dont write anything else yet.
Just included the lib header and copy paste the rest just to see if it is
compile or not. But it is not. :(
I wanted to communicate with a PLC via modbus with this lib.

Bert Melis <notifications@github.com> ezt írta (időpont: 2019. aug. 14.,
Sze 9:20):

zekageri, 2019-08-14:
Yes, i have got some more errors but i think all of it are related to this error. 
Other error messages are two include errors. the stdint.h and the functional.
I don't know why these are here. I'am using PlatformIO ide with vscode and latest esp32 core in it with latest async lib.

bertmelis, 2019-08-14:
You mean you cannot include stdint and functional? Well, if these things are not included, the file will not compile.

Surprising however because these files belong to the Arduino core and should be available when installign through Platformio.

zekageri, 2019-08-14:
They are included just the platformIO can not open them for some reason. I use my own library folder inside my project. Every library is in that folder. And some of them are using these libs too without a problem. :/

closed 2020-05-06 by bertmelis