bertmelis/esp32ModbusTCP #7: Debug problem

HilverinkJ, 2019-09-23:
Hello Bert,
I'm triying to debug your example code. My modbus server is a Eastron ESP 2341 modbus TCP/RTU gateway. 
Debugging is hanging in this part of code. I can't find/do not understand why.

Do you have any idea?

Regards,
Joop Hilverink

[image]

bertmelis, 2019-09-24:
I haven't used the debugger yet and never inspected assembler (asm knowledge almost unexsting).

From your screen I don't know at which point of the program it halts.

HilverinkJ, 2019-09-25:
Can you advice how to check where it is going wrong? slave 1 (SDM220) is showing activety on the screen.
[image]

This is the serial output:

> 
--- Miniterm on COM8  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5800
entry 0x4008069c
�
Connecting to WiFi...
WiFi connected. IP: 192.168.178.68

reading registers
power: 0
energy: 0
>

bertmelis, 2019-09-25:
You should build with debug enabled: add -D CONFIG_LOG_DEFAULT_LEVEL=ESP_LOG_DEBUG to your building script (set flags in platformio.ini?)

See https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/log.html