bertmelis/esp32ModbusTCP #10: SMA.ino with a SMA inverter - getting error e4 for each register read

planB-dev, 2019-11-06:
Hey Bert,

First, thank you so much for this project! This is exactly what I was trying to achieve. I'm running exactly this same setup as you ESP32 board + SMA inverter on a LAN.

I have succes to access the registers of the SunnyBoy SMA inverter manually from a PC-based modbus client on the same LAN, but I systematically get errors while using SMA.ino.

- It successfully connects to my SSID and LAN.
- It tries to read registers, and gets error E4
- it then crashes and reboot:

---------------------------
_Connecting to WiFi... 
WiFi connected. IP: 192.168.10.151
reading registers
Error packet 1: e4
Error packet 2: e4
Error packet 3: e4
Error packet 4: e4
Error packet 5: e0
Error packet 6: e0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4013b7ab  PS      : 0x00060f30  A0      : 0x800d2c84  A1      : 0x3ffcf5d0  
A2      : 0x3ffcf5f8  A3      : 0x0000000c  A4      : 0x0000000c  A5      : 0x00000000_  

---------------------------

I tries to change and explore a few different register addresses with the same result. But they all work ok with a PC-client.

Have you ever experienced this?

Thanks Again,
Pascal

bertmelis, 2019-11-06:
I'm sure I have had this error. Could you capture the rest of the dump? (and do a stack trace decode?) The crash is probably related to an invalid pointer.

Do the errors come immediately or only after some time (seconds).

Unfortunately, I don't have any modbus devices anymore so I cannot test. (unless I setup a simulator but it seems like a chore and I'm postponing it untill eternity, sorry)

planB-dev, 2019-11-08:
Hi Bert,

thanks for the quick feedback! 

Here is the complete stack dump:
___________
_Connecting to WiFi... 
WiFi connected. IP: 192.168.10.184
reading registers
Error packet 1: e4
Error packet 2: e4
Error packet 3: e4
Error packet 4: e4
Error packet 5: e4
Error packet 6: e4
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4013b7ab  PS      : 0x00060030  A0      : 0x800d2c84  A1      : 0x3ffcff00  
A2      : 0x3ffcff28  A3      : 0x0000000c  A4      : 0x0000000c  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x3ffb1f90  A8      : 0x00000000  A9      : 0x3ffcfec0  
A10     : 0x00000000  A11     : 0x3ffcfeec  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x4013b7ab:0x3ffcff00 0x400d2c81:0x3ffcff20 0x4013b8c5:0x3ffcff60 0x400d3469:0x3ffcff80 0x400d34c1:0x3ffcffc0 0x400d391e:0x3ffcffe0 0x400883d1:0x3ffd0010

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)_
___________

I'm having trouble to getting the stack decoder tool to work.

The errors happens after each of the 6 register reading (as per trace). It's an immediate problem. 

I tried connecting to the Inverter with PC or android-based client as a control test, it works.

If you have any idea of what causes error "4", let me know. I would have thought it might be a modbus-TCP protocol variant issue, but since your code was written and tested on the same SMA inverter family, I doubt this is it.

Thanks.

ReneHezser, 2020-03-06:
Is there an update to the problem? I am receiving the same error :-(

alfmalo000, 2020-03-23:
Try to reboot the sunny boy o reboot the TCP and UDP modbus on configuration sunny boy.

bertmelis, 2020-03-26:
I'm sorry to have you guys waiting. But now I'm stuck at home I'll try to find some time to look into this entire lib (in between supporting home school).

fprumbau, 2020-05-03:
Having the same error, I decoded the stacktrace from above as

PC: 0x4013b7ab: esp_ota_get_running_partition at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/app_update/esp_ota_ops.c line 510
EXCVADDR: 0x00000000

Decoding stack results
0x4013b7ab: esp_ota_get_running_partition at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/app_update/esp_ota_ops.c line 510
0x400d2c81: AsyncClient::_close() at /home/fprumbau/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp line 815
0x4013b8c5: esp32ModbusTCPInternals::ModbusResponse::getSlaveAddress() at /tmp/arduino_build_997696/sketch/ModbusMessage.cpp line 202
0x400d3469: HardwareSerial::end() at /home/fprumbau/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/HardwareSerial.cpp line 88
0x400d34c1: HardwareSerial::begin(unsigned long, unsigned int, signed char, signed char, bool, unsigned long) at /home/fprumbau/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/HardwareSerial.cpp line 50
0x400d391e: String::changeBuffer(unsigned int) at /home/fprumbau/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/WString.cpp line 179
0x400883d1: prvInitialiseNewQueue at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 430

bertmelis, 2020-05-03:
Could you try again with the v1 branch?

fprumbau, 2020-05-04:
Wow, thank you for looking into this that quickly :-)

With v1 branch the crashes have gone for me.

It does read some values now, so it is working in principle!

_Connecting to WiFi... 
WiFi lost connection. Reason: 202
WiFi connected. IP: 192.168.178.52
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51
currentdc1power: 551
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51
currentdc2power: 91
reading registers:
  requested status
  requested connectionstatus
  requested totalpower
  requested currentpower
  requested currentdc1power
  requested currentdc2power
status: 307
connectionstatus: 51_

fprumbau, 2020-05-04:
Maybe it's some timing issue. I modified the registers to only three entries. I am always getting 2 values. For me, that's OK (because the 3rd could be calculated).

_reading registers:   requested currentpower,  requested currentdc2power,  requested currentdc1power,
currentpower: 1512
currentdc2power: 120

reading registers:   requested currentpower,  requested currentdc2power,  requested currentdc1power,
currentpower: 1528
currentdc2power: 121

reading registers:   requested currentpower,  requested currentdc2power,  requested currentdc1power,
currentpower: 1568
currentdc2power: 121

reading registers:   requested currentpower,  requested currentdc2power,  requested currentdc1power,
currentpower: 1638
currentdc2power: 121

reading registers:   requested currentpower,  requested currentdc2power,  requested currentdc1power,
currentpower: 2206
currentdc2power: 123
_

bertmelis, 2020-05-04:
The v1 version sends all requests without waiting for the previous to complete. It could be the SMA is not capable of handling all the requests. I (temporarily) moved so I cannot test myself anymore.

A workaround could be to only send your next request when the previous completes. Do this by raising a flag and request outside of the callback. (my computer is occupied by home schooling kids, I can't check code for race issues or deadlocks)