bertmelis/esp32ModbusTCP #11: Writing data in to a holding register.

MaxAbarca1993, 2020-01-15:
Hi Bert, I'm so iterested on this library, it worked very well for  me, but i don't now if it has implemented  the write holding regiters yet. I really hope if you can help me with this. I'm working on a water well system that needs to send data via modbus tcp and store it on a database, so, the data acquirer is a ESP32 board. Thank you very much for your work, it was a great advace for my proyect.

berguinsson, 2020-03-12:
Hi MaxAbarca1993, 
Did you get something? I have to do something similar as you. 
I'm looking forward to hearing from you!

sigitnurmajid, 2020-04-22:
Hi MaxAbarca1993,
Did you get something? I have to do something similar as you.
I'm looking forward to hearing from you!

bertmelis, 2020-04-22:
I'm writing a completely new version. The current TCP management has a few flaws. I'll upload the new version this evening (in a new branch). API stays the same. It'll include FC03, write holding registers. (I already have FC06, write single register)

bertmelis, 2020-04-22:
Please have a look at branch [v1](https://github.com/bertmelis/esp32ModbusTCP/tree/v1).

I was a bit confused earlier. Do you need FC06 (write holding register) or FC16 (set multiple registers).

MaxAbarca1993, 2020-04-22:
Hi bertmelis, first of all, thanks for your work and your answer. About your doubt, I needed FC06 write holding registers for my project.

bertmelis, 2020-04-23:
Ok, FC06 is included in the v1-branch. Could you try this? I'm not able to test in various situations because I don't have any modbus devices anymore.

sigitnurmajid, 2020-04-24:
Hi bertmelis, I have already use your code to read and write data to holding register and it's works but I can't set my cycle time for read and write less than 1 second and status modbus always connecting.

bertmelis, 2020-04-24:
You mean you read and write every second? I'll try some stress testing myself, but that's pretty fast for modbus.

So it works but not if yo read/write too fast?

sigitnurmajid, 2020-04-24:
I have project to controll and monitoring some modbus device, and I need to read and write registers about 50ms/looping to chasing realtime. Is it possible to use modbus protocol?

`   lastMilis = 0;
    if ((millis() - lastMillis > 50)) {
    lastMillis = millis();
    MB.readHoldingRegisters(address, length)
  }`

Here code for example what i want.

bertmelis, 2020-04-24:
Maybe it is possible but I wouldn't use modbus to poll a device every 50ms. May I ask what kind of project you're building?