bertmelis/esp32ModbusTCP #13: Bad message response

pablozg, 2020-04-28:
Hi, I'm testing your new V1 code, but I get same data payload in any address I query.

After check the changes I see this in esp32ModbusMessage.cpp:

uint8_t* ModbusResponse::getData() {
  return &_buffer[6];
}

If I change the buffer index to 9 I start to get the correct payload again.

uint8_t* ModbusResponse::getData() {
  return &_buffer[9];
}

Thanks for you work.

bertmelis, 2020-04-28:
Oops...

rectified in https://github.com/bertmelis/esp32ModbusTCP/commit/b37a4ba88c8dd343b6a60a06093d1c6bd31a6c4b

pablozg, 2020-05-01:
Hi again, I'm having a lot of reboots with the new version, but I can figure what happen.

Here is some reboots and the decoding results:

````
"01/05/2020 11:14:41",[V][esp32ModbusTCP.cpp:133] _tryToSend(): send id 1974
"01/05/2020 11:14:43",abort() was called at PC 0x4016b64f on core 1

"01/05/2020 11:14:43",Backtrace: 0x4008cde0:0x3ffb1d20 0x4008d011:0x3ffb1d40 0x4016b64f:0x3ffb1d60 0x4016b696:0x3ffb1d80 0x4016b2f1:0x3ffb1da0 0x4016b074:0x3ffb1dc0 0x4016b039:0x3ffb1de0 0x400ee879:0x3ffb1e00 0x400ee889:0x3ffb1e20 0x400ee8c3:0x3ffb1e40 0x400ee011:0x3ffb1e60 0x400d5de3:0x3ffb1e80 0x400d5e86:0x3ffb1eb0 0x400e53d3:0x3ffb1ed0 0x400e53fb:0x3ffb1ef0 0x400f74cd:0x3ffb1f10 0x400f76af:0x3ffb1f40 0x400e164d:0x3ffb1f80 0x400fbdb1:0x3ffb1fb0 0x40089479:0x3ffb1fd0

"01/05/2020 11:14:43",Rebooting...
"01/05/2020 11:14:43",ets Jun  8 2016 00:22:57

Decoding stack results
0x4008cde0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008d011: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4016b64f: std::string::_Rep::_M_destroy(std::allocator  const&) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/ext/new_allocator.h line 110
0x4016b2f1: __gnu_cxx::__scoped_lock::~__scoped_lock() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/ext/concurrence.h line 246
0x4016b074: __cxxabiv1::__cxa_rethrow() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 93
0x4016b039: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 71
0x400ee879: ModbusRequest03::~ModbusRequest03() at lib\esp32ModbusTCP\src/esp32ModbusMessage.h line 74
0x400ee889: ModbusMessage::ModbusMessage(unsigned int) at lib\esp32ModbusTCP\src\esp32ModbusMessage.cpp line 80
0x400ee8c3: ModbusMessage::ModbusMessage(unsigned int) at lib\esp32ModbusTCP\src\esp32ModbusMessage.cpp line 85
0x400ee011: esp32ModbusTCP::readHoldingRegisters(unsigned char, unsigned short, unsigned short, void*) at lib\esp32ModbusTCP\src\esp32ModbusTCP.cpp line 97
0x400d5de3: victron() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/modbustcp.ino line 241
0x400d5e86: readMeter() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/modbus.ino line 310
0x400e53d3: getSensorData() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/Support_functions.ino line 33
0x400e53fb: getSensorData() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/Support_functions.ino line 54
0x400f74cd: _GLOBAL__sub_I__ZN10SPIFFSImplC2Ev() at C:\.platformio\packages\framework-arduinoespressif32\cores\esp32/IPAddress.h line 94
0x400f76af: TickerScheduler::enable(unsigned char) at lib\TickerScheduler\TickerScheduler.cpp line 96
0x400e164d: loop() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/FreeDS.ino line 802
0x400fbdb1: base64_encode_block at C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\libb64\cencode.c line 55
0x40089479: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

"01/05/2020 11:26:01",[V][esp32ModbusTCP.cpp:133] _tryToSend(): send id 2004
"01/05/2020 11:26:01",Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
"01/05/2020 11:26:01",Core 0 register dump:
"01/05/2020 11:26:01",PC      : 0x400e637f  PS      : 0x00060030  A0      : 0x80150515  A1      : 0x3ffb4e20  
"01/05/2020 11:26:01",A2      : 0x3ffdf15c  A3      : 0x3ffde2c8  A4      : 0x00000005  A5      : 0x00000020  
"01/05/2020 11:26:01",A6      : 0x00000020  A7      : 0x00000000  A8      : 0x00000004  A9      : 0x3ffb4e00  
"01/05/2020 11:26:02",A10     : 0x00000000  A11     : 0x40085804  A12     : 0x3ffb9964  A13     : 0x00000000  
"01/05/2020 11:26:02",A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
"01/05/2020 11:26:02",EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

"01/05/2020 11:26:02",Backtrace: 0x400e637f:0x3ffb4e20 0x40150512:0x3ffb4e50 0x401509de:0x3ffb4e70 0x40151517:0x3ffb4e90 0x401513f9:0x3ffb4eb0 0x40151604:0x3ffb4ed0 0x401497fc:0x3ffb4ef0 0x40089479:0x3ffb4f20

"01/05/2020 11:26:02",Rebooting...
"01/05/2020 11:26:02",ets Jun  8 2016 00:22:57

PC: 0x400e637f
EXCVADDR: 0x00000000

Decoding stack results
0x40150512: tcp_slowtmr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1175
0x401509de: tcp_process_refused_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1383
0x40151517: sys_timeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 227
0x401513f9: tcp_netif_ip_addr_changed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 2127
0x40151604: tcpip_tcp_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 144
[... cut ...]
0x400f74cd: _GLOBAL__sub_I__ZN10SPIFFSImplC2Ev() at C:\.platformio\packages\framework-arduinoespressif32\cores\esp32/IPAddress.h line 94
0x400f76af: TickerScheduler::enable(unsigned char) at lib\TickerScheduler\TickerScheduler.cpp line 96
0x400e164d: loop() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/FreeDS.ino line 802
0x400fbdb1: base64_encode_block at C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\libb64\cencode.c line 55
0x40089479: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

````

bertmelis, 2020-05-01:
Is it possible to include a bit more from your logging? I'd like to know what happens before eg. in what state the lib is in.

pablozg, 2020-05-01:
Hi, I add some log lines to see the variable status and try to find any corrupt data or buffer overflow, if you need more lines I will record the log again. thanks.

functions changed:

````
ModbusMessage::ModbusMessage(size_t totalLength) :
  _buffer(nullptr),
  _totalLength(totalLength),
  _index(0) {
    log_v("totalLength %d", totalLength);
    log_v("_totalLength %d", _totalLength);
    if (_totalLength > 256) _totalLength = 256;  // max Modbus length message
    log_v("_totalLength %d", _totalLength);
    log_v("_buffer %d", _buffer);
    _buffer = new uint8_t[_totalLength];
  }

ModbusRequest::ModbusRequest(size_t totalLength) :
  ModbusMessage(totalLength),
  _packetId(0),
  _slaveAddress(0),
  _functionCode(0),
  _address(0),
  _byteCount(0) {
    _packetId = ++_lastPacketId;
    log_v("_packetId %d", _packetId);
    log_v("_lastpacketId %d", _lastPacketId);
    if (_lastPacketId == 0) _lastPacketId = 1;
  }

This function was changed because I need to read some variables in two differents id's
(victron inverters have different id for battery, inverter, bms, etc) so I open a connection 
to the server and I pass to the function the right id and the others params.
 (with the previous version works without issues, I can send a PR if you like the idea).

uint16_t esp32ModbusTCP::readHoldingRegisters(uint8_t serverId, uint16_t address, uint16_t numberRegisters, void* arg) {
  log_v("serverId %d address %d numberRegisters %d", serverId, address, numberRegisters);
  ModbusRequest* request = new ModbusRequest03(serverId, address, numberRegisters);
  if (request) return _addToQueue(request, arg);
  return 0;

````

And the log result:

````
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 100 address 842 numberRegisters 1
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 1992
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 1992
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:134] _tryToSend(): send id 1987
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:134] _tryToSend(): send id 1988
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:134] _tryToSend(): send id 1989
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:134] _tryToSend(): send id 1990
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:134] _tryToSend(): send id 1991
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:134] _tryToSend(): send id 1992
"01/05/2020 12:45:57",[V][esp32ModbusTCP.cpp:233] _onData(): data: len 11
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 11
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 11
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 11
"01/05/2020 12:45:57",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 12:45:59",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 9 numberRegisters 1
"01/05/2020 12:45:59",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 12:45:59",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 12:45:59",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 12:45:59",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 12:45:59",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 1993
[... 236 lines cut ...]
"01/05/2020 12:56:57",Rebooting...
"01/05/2020 12:56:57",ets Jun  8 2016 00:22:57

Decoding stack results
0x4008cde0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008d011: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4016b727: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x4016b76e: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4016b3c9: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x4016b14c: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x4016b111: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x400ee8fd: ModbusMessage::ModbusMessage(unsigned int) at lib\esp32ModbusTCP\src\esp32ModbusMessage.cpp line 89
0x400ee90e: ModbusRequest::ModbusRequest(unsigned int) at lib\esp32ModbusTCP\src\esp32ModbusMessage.cpp line 104
0x400ee99b: ModbusRequest03::ModbusRequest03(unsigned char, unsigned short, unsigned short) at lib\esp32ModbusTCP\src\esp32ModbusMessage.cpp line 137
0x400ee03f: esp32ModbusTCP::readHoldingRegisters(unsigned char, unsigned short, unsigned short, void*) at lib\esp32ModbusTCP\src\esp32ModbusTCP.cpp line 98
0x400d5e0b: victron() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/modbustcp.ino line 245
[... cut ...]
0x400f75a5: TickerScheduler::handleTicker(std::function , void*, bool volatile*) at c:\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400f7787: TickerScheduler::update() at lib\TickerScheduler\TickerScheduler.cpp line 132
0x400e1675: loop() at E:/Programacion/FreeDs/FreeDS Modbus Library V1/src/FreeDS.ino line 807
0x400fbe89: loopTask(void*) at C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\main.cpp line 19
0x40089479: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
````

bertmelis, 2020-05-01:
Including the server ID in the request is indeed useful. It enables communication via gateways. So a PR is welcome. Maybe implement an overloaded method that includes the ID so you can have it both ways. (I assume it will be optimized out of final binary if not used.)

I'm thinking of a memory leak somewhere because it's a store error. I haven't implemented testing (yet) in this branch... How's your free heap?

bertmelis, 2020-05-01:
Aaaaa. The struct `ModbusAction` doesn't contain a destructor so it leaks the request and response.

It should also contain a cope/move constructor but I'm figuring out if this is really necessary.

you could already try this:

```C++
  struct ModbusAction {
    ModbusAction(ModbusRequest* req, ModbusResponse* resp, void* a):
      request(req),
      response(resp),
      arg(a) {}
    ~ModbusAction() {
      delete request;
      delete response;
    }
    ModbusRequest* request;
    ModbusResponse* response;
    void* arg;
  };
```
in esp32ModbusTCP.h

pablozg, 2020-05-01:
yes, of course, thanks.

pablozg, 2020-05-01:
don't work, restart after a few request

````
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 12 numberRegisters 1
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 210476 bytes
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 2
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 2
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 26 numberRegisters 1
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 210372 bytes
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 3
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 3
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 27 numberRegisters 1
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 210268 bytes
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 4
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 4
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 61 numberRegisters 1
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 210164 bytes
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 5
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 5
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 100 address 842 numberRegisters 1
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 210260 bytes
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:187] _onConnect(): connected
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 6
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 6
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 1
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 2
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 3
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 4
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 5
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 6
"01/05/2020 14:44:33",[V][esp32ModbusTCP.cpp:234] _onData(): data: len 11
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 11
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 11
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 11
"01/05/2020 14:44:33",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 14:44:33",CORRUPT HEAP: Bad head at 0x3ffd0234. Expected 0xabba1234 got 0x3ffd0390
"01/05/2020 14:44:33",assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
"01/05/2020 14:44:33",abort() was called at PC 0x40135b0b on core 1

"01/05/2020 14:44:33",Backtrace: 0x4008cde0:0x3ffd40b0 0x4008d011:0x3ffd40d0 0x40135b0b:0x3ffd40f0 0x4008ca55:0x3ffd4120 0x4008540e:0x3ffd4140 0x40085815:0x3ffd4160 0x4000bec7:0x3ffd4180 0x4016b129:0x3ffd41a0 0x4016b4b1:0x3ffd41c0 0x400ee833:0x3ffd41e0 0x400ee851:0x3ffd4200 0x4017a82b:0x3ffd4220 0x400ee3fe:0x3ffd4240 0x400ee669:0x3ffd4260 0x4017a801:0x3ffd4290 0x400e6b11:0x3ffd42b0 0x400e6ba5:0x3ffd42f0 0x400e72a9:0x3ffd4310 0x40089479:0x3ffd4340

"01/05/2020 14:44:33",Rebooting...
"01/05/2020 14:44:33",ets Jun  8 2016 00:22:57

[... cut ...]
0x400e6b11: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at c:\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400e6ba5: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at lib\AsyncTCP\src\AsyncTCP.cpp line 1210
0x400e72a9: _async_service_task(void*) at lib\AsyncTCP\src\AsyncTCP.cpp line 162
0x40089479: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
````

bertmelis, 2020-05-01:
Yeah, we also need move and copy constructor...

pablozg, 2020-05-01:
Ok, after comment out the lines, still working, so I'm logging the output to see what happens, in this time I do a new test to see the free heap value along the time:

- First crash
````
"01/05/2020 14:49:27",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 203780 bytes

occurs with 

"01/05/2020 15:00:11",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 58352 bytes
````

- Second crash 

````
"01/05/2020 15:00:25",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 211376 bytes

occurs with

"01/05/2020 15:01:43",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 134096 bytes
````

third crash 

''''
"01/05/2020 15:01:57",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 211356 bytes

occurs with

"01/05/2020 15:12:25",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 57284 bytes
''''

and now with the suggested changes (modbusAction)

'''''
"01/05/2020 15:25:18",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 169060 bytes

occurs with

"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 26 numberRegisters 1
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 58036 bytes
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 1963
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:135] _tryToSend(): send id 1964
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 1965
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 1965
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 27 numberRegisters 1
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 57928 bytes
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 1966
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 1966
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 239 address 61 numberRegisters 1
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 57824 bytes
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:106] ModbusRequest(): _packetId 1967
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:107] ModbusRequest(): _lastpacketId 1967
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:97] readHoldingRegisters(): serverId 100 address 842 numberRegisters 1
"01/05/2020 15:33:23",[V][esp32ModbusTCP.cpp:98] readHoldingRegisters(): Free Heap: 57720 bytes
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:84] ModbusMessage(): totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:85] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:87] ModbusMessage(): _totalLength 12
"01/05/2020 15:33:23",[V][esp32ModbusMessage.cpp:88] ModbusMessage(): _buffer 0
[... 7 lines cut ...]
"01/05/2020 15:33:23",A10     : 0x00000000  A11     : 0x40085804  A12     : 0x3ffbb2d0  A13     : 0x00000000  
"01/05/2020 15:33:23",A14     : 0x3fffece8  A15     : 0x3fffece8  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
"01/05/2020 15:33:23",EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

"01/05/2020 15:33:23",Backtrace: 0x400e637f:0x3ffb4e20 0x401505fa:0x3ffb4e50 0x40150ac6:0x3ffb4e70 0x401515ff:0x3ffb4e90 0x401514e1:0x3ffb4eb0 0x401516ec:0x3ffb4ed0 0x401498e0:0x3ffb4ef0 0x40089479:0x3ffb4f20

"01/05/2020 15:33:23",Rebooting...
"01/05/2020 15:33:23",ets Jun  8 2016 00:22:57

PC: 0x400e637f: _tcp_poll(void*, tcp_pcb*) at lib\AsyncTCP\src\AsyncTCP.cpp line 259
EXCVADDR: 0x00000000

Decoding stack results
0x400e637f: _tcp_poll(void*, tcp_pcb*) at lib\AsyncTCP\src\AsyncTCP.cpp line 259
0x401505fa: tcp_slowtmr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1213
0x40150ac6: tcp_tmr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 164
0x401515ff: tcpip_tcp_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 140
0x401514e1: sys_check_timeouts at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 361
0x401516ec: sys_timeouts_mbox_fetch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 433
0x401498e0: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 109
0x40089479: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

'''''

I'm sorry but my c++ skills are limited.

bertmelis, 2020-05-01:
No problem. I'm diving in. Setting up my modbus simulator...

For the record: do you actually get responses?

pablozg, 2020-05-01:
Yes, the data still working, the only issue at the moment is the continuously restarts.

bertmelis, 2020-05-01:
I've made some changes. Could you try again with the updated code?

pablozg, 2020-05-01:
Hi, now the free head seems much stable

````
17:48:44 - Free Heap: 202888 bytes
17:48:56 - Free Heap: 202684 bytes
17:48:57 - Free Heap: 202328 bytes
17:48:58 - Free Heap: 202900 bytes
17:48:59 - Free Heap: 202796 bytes
17:49:01 - Free Heap: 202888 bytes
17:49:02 - Free Heap: 202884 bytes
17:49:03 - Free Heap: 202480 bytes
17:49:04 - Free Heap: 202892 bytes
17:49:05 - Free Heap: 202656 bytes
17:49:06 - Free Heap: 202896 bytes
17:49:07 - Free Heap: 202896 bytes
17:49:08 - Free Heap: 192024 bytes
17:49:09 - Free Heap: 202896 bytes
17:49:11 - Free Heap: 202900 bytes
17:49:12 - Free Heap: 202912 bytes
17:49:13 - Free Heap: 202896 bytes
17:49:14 - Free Heap: 202900 bytes
17:49:15 - Free Heap: 202892 bytes
17:49:15 - Free Heap: 202884 bytes
17:49:16 - Free Heap: 202900 bytes
17:49:17 - Free Heap: 200796 bytes
17:49:19 - Free Heap: 202896 bytes
17:49:20 - Free Heap: 190716 bytes
17:49:21 - Free Heap: 202300 bytes
17:49:22 - Free Heap: 202900 bytes
17:49:23 - Free Heap: 202616 bytes
17:49:24 - Free Heap: 202896 bytes
17:49:25 - Free Heap: 202900 bytes
17:49:26 - Free Heap: 202896 bytes
17:49:27 - Free Heap: 200816 bytes
17:49:28 - Free Heap: 202900 bytes
17:49:29 - Free Heap: 202368 bytes
17:49:30 - Free Heap: 202888 bytes
17:49:31 - Free Heap: 202896 bytes
````

now I'm testing the new changes, I will wait some time and see if everything is working well. Thanks.

P.D. I sent you a PR with my changes and a new example using a victron.

pablozg, 2020-05-02:
Hi, after more of 13 hours without restarts, I think we can say that the issue is fixed. Thanks a lot.

bertmelis, 2020-05-02:
Thank you for testing. I'll move V1 to master shortly. And merge your PR...

closed 2020-05-02 by bertmelis