bertmelis/esp32ModbusTCP #16: Problems with connection and comunication!

marzintoshtech, 2020-05-06:
Good morning dear Bert.
I am writing to you first of all to congratulate you for all the excellent work you do!
I am writing to you because I have an exigency!
I am designing an electronic board to which 2 Ds18b20 sensors, an alternating current sensor, digital inputs and outputs are connected. 
I have to transmit the temperature, inputs, current, digital i / o data to a remote supervisor via the company wifi network.
All this I need for my company!
I tried to use your library but it always gives me communication errors! 
I am using as master on the PC a library for C # called EasyModbusTCP .NET Package V5.5.
I can't open the connection nor send data! 

Could you help me, even for a **fee**?

Thanks a lot by Davide

bertmelis, 2020-05-06:
Does your PC's firewall allow incoming connections? Could you try to make a connection with another PC? (eg using https://packetsender.com/)

bertmelis, 2020-05-06:
Oh, and are you using branch v1? Or master?

marzintoshtech, 2020-05-06:
I use esp32ModbusTCP-master. 
version = 0.0.1

> Does your PC's firewall allow incoming connections? Could you try to make a connection with another PC? (eg using https://packetsender.com/).
> 
> Edit: also try to build the code with debug enabled for AsyncTCP and watch the Serial interface to see what's happening.
> Are you on Arduino IDE or Platformio or...?

I use Arduino ide

marzintoshtech, 2020-05-06:
> Oh, and are you using branch v1? Or master?

I use esp32ModbusTCP-master. 
version = 0.0.1

marzintoshtech, 2020-05-06:
Before I used esp8266 and I managed to make the library and communication work!
While for the esp32 modbus I don't know where to start!
Please, can you help me?

bertmelis, 2020-05-07:
I can help to a certain extent. But it will be easier if you use VSCode with Platformio to build your firmware.

Second, if the lib can't make a connection, there is something wrong outside of this lib.

So the sketch in your repo has the be rewritten for esp32?

bertmelis, 2020-05-07:
The repo in your link is a modbus slave aka server. This library is to make a modbus master aka client.

marzintoshtech, 2020-05-07:
> Posso aiutare fino a un certo punto. Ma sarà più semplice se usi VSCode con Platformio per costruire il tuo firmware.
> 
> Secondo, se la lib non riesce a stabilire una connessione, c'è qualcosa di sbagliato al di fuori di questa lib.
> 
> Quindi lo schizzo nel tuo repository deve essere riscritto per esp32?

Hi Bert!
Thank you very much for your quick reply! You are very very kind to dedicate your time to me!
I installed the software you indicated to me, that is VSCode with Platformio.
I did not use the sketches I sent you on the ESP32(ESP8266WiFi.h) because it immediately gave an error during compilation! I used to test the example in your esp32ModbusTCP library, the example called SMA.ino. But not work with Modbus simulator!
I used this masterModbus to try and communicate:
MODBUS SIMULATOR from Calta computer System limited.

thanks for your patience

bertmelis, 2020-05-07:
But you are aware this library only implements a modbus master? So your simulator should be in slave mode. Next, make sure your computer is accessible (eg, the firewall allows traffic coming through).

If that is the case, build the firmware with debug enabled. add this to your platformio.ini

```
build_flags =
  -Wall
  -DCORE_DEBUG_LEVEL=5
```
 Then, post the debug log (Serial ouput).

bertmelis, 2020-05-08:
You got me thinking... If you need a modbus slave. I can make a raw one for you. Let me know.

marzintoshtech, 2020-05-08:
> Mi fai pensare ... Se hai bisogno di uno schiavo modbus. Posso farne uno di base per te. Fammi sapere.

Dear Bert! Thank you very much for the time you dedicate to me! I'd be really happy if you would make me a basic one!
But I don't want to take advantage of your generosity!
I would like to send you a compensation from my company, through Paypal!
I'll explain in detail what I need to do for the company I work with:
I am planning a wifi device with esp32 that will be mounted in every convector present in the administrative building of my company! There are about 170 convectors!
The electronic board I am designing will control 2/3 DS18B20 sensors, 4 analog inputs, 8 digital inputs, an analog input from an amperometric transformer, 4 relay outputs.
I would like to transmit these analog and digital values ​​via modbus tcp / ip over the company wifi network or an ad hoc network.
All electronic boards will be queried cyclically one at a time by a software created by a colleague in C #. As a basis it will use the "EasyModbus.dll" library.
Since we have been standing for more than two months, my bosses have rushed me, not knowing what it means to develop something new! They only think about numbers and earnings!
I had managed to create something with ESP8266, but having few inputs, I wanted to opt for the ESP32 which has multiple analog and digital inputs and also has bluetooth!
I'm sorry to bother you and ask you for help but I'm a pure electronic designer who adapts to being a software engineer! Unfortunately I have never worked with modbus! In the past I was an assembler and I find myself in difficulty in today's world where MCU software is developed with the C language.
I'm so sorry for dwelling on it!

You are very kind with me! Thanks for your valuable help! God bless you!

marzintoshtech, 2020-05-08:
[EasyModbusLibrary for .NET (DLL).zip](https://github.com/bertmelis/esp32ModbusTCP/files/4598613/EasyModbusLibrary.for.NET.DLL.zip)
[EasyModbus Server Simulator (.NET Version).zip](https://github.com/bertmelis/esp32ModbusTCP/files/4598615/EasyModbus.Server.Simulator.NET.Version.zip)
[EasyModbusClient (.NET Version).zip](https://github.com/bertmelis/esp32ModbusTCP/files/4598617/EasyModbusClient.NET.Version.zip)

bertmelis, 2020-05-08:
Well, I can probably create a modbus server. But I'm not a professional programmer and it is in my free time. So don't expect me to adhere to a deadline or make robust and bug free code.

Anyway, I'll have a look tonight. Will push what I have into a separate repo and give the link here. From there on we'll see how it goes.

bertmelis, 2020-05-08:
For the record. ESP32 has a modbus library builtin. Look in the ESP-IDF repo.

marzintoshtech, 2020-05-09:
> Well, I can probably create a modbus server. But I'm not a professional programmer and it is in my free time. So don't expect me to adhere to a deadline or make robust and bug free code.
>
> Anyway, [...]

Thanks a lot Bert! Have you a nice weekend!!

bertmelis, 2020-05-11:
I'm still breaking my head over a message parser. So I may need a few days to finish...

But if you want to get your application running, you should seriously consider using the ESP-IDF framework. It has a built-in modbus library and ready to use drivers for GPIO.

marzintoshtech, 2020-05-11:
> ESP-IDF framework

Hi Bert! OK, i try it! Don't worry about creating the modbus server for me! I will try to use ESP-IDF framework!
Thanks so much for taking the time!

bertmelis, 2020-05-11:
Gonna keep the topic open. I'm already working on it so now it's a matter of dignity to finish it.

marzintoshtech, 2020-05-11:
> Gonna keep the topic open. I'm already working on it so now it's a matter of dignity to finish
Sorry Bert! 
It was not my intention to offend you! 
**Forgive me!** 
I thought you didn't want to go on with the project anymore! 
Do not get angry! Sorry again!

bertmelis, 2020-05-11:
Euh...

I was just saying that if you have a deadline and your code has to be watertight, you should look into the ESP-IDF framework. For modbus, that is definitely faster.

Simple GPIO is also easy. And I even don't know if ADC is already implemented in the Arduino core. So you'll have to dig into that framework anyway.

It's C, not C++ but it has great documentation.

marzintoshtech, 2020-05-11:
> Euh ...
> 
> Stavo solo dicendo che se hai una scadenza e il tuo codice deve essere a tenuta stagna, dovresti esaminare il framework ESP-IDF. Per modbus, questo è decisamente più veloce.
> 
> Anche GPIO semplice è facile. E non so nemmeno se ADC sia già implementato nel core di Arduino. Quindi dovrai scavare in quel quadro comunque.
> 
> È C, non C ++ ma ha un'ottima documentazione.

The problem is that I don't know where to start with the ESP-IDF framework. 
I can't learn another programming language in a short time! I will manage with arduino language! The application is not critical or for security use, it is only for monitoring.
By the way, PlatformIO, I got to try it and it is much more powerful than the standard arduino IDE. I Thank you for letting me know! Good evening dear Bert! **Best regards from Italy!!**

bertmelis, 2020-05-14:
Check out https://github.com/bertmelis/espModbus/tree/develop

Only FC03 implemented though. Trying something else completely undefined behaviour. And it needs a lot of testing...

I'm closing here. Feel free to create an issue in the other repo for further discussion.

closed 2020-05-14 by bertmelis

marzintoshtech, 2020-05-18:
> Dai un'occhiata a https://github.com/bertmelis/espModbus/tree/develop
> 
> Tuttavia, solo FC03 è stato implementato. Provare qualcos'altro comportamento completamente indefinito. E ha bisogno di molti test ...
> 
> Sto chiudendo qui. Sentiti libero di creare un problema nell'altro repository per ulteriori discussioni.

Thank's so much Bert! I will try this software!
Best regards