bertmelis/esp32ModbusTCP #17: Load prohibited on tcp timeout

pablozg, 2020-05-28:
Hi, sometimes I have an issue when tcp timeout occurs, 

````

14:27:31.340 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 593
14:27:31.807 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 13
14:27:32.043 ->   requested 40079
14:27:32.327 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 594
14:27:32.990 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 13
14:27:33.037 ->   requested 40071
14:27:33.320 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 595
14:27:33.976 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 13
14:27:34.023 ->   requested 499
14:27:34.355 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 596
14:27:35.060 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 21
14:27:35.532 ->   requested 40265
14:27:35.813 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 597
14:27:39.298 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 17
14:27:39.536 ->   requested 40282
14:27:39.824 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 598
14:27:40.728 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 15
14:27:40.822 -> [W][AsyncTCP.cpp:962] _poll(): ack timeout 4
14:27:41.013 ->   requested 40302
14:27:41.341 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 599
14:27:46.389 -> [W][AsyncTCP.cpp:962] _poll(): ack timeout 4
14:27:46.389 -> [V][esp32ModbusTCP.cpp:263] _onTimeout(): TCP timeout
14:27:46.389 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
14:27:46.389 -> Core 1 register dump:
14:27:46.389 -> PC      : 0x40177b8f  PS      : 0x00060a30  A0      : 0x800ee051  A1      : 0x3ffdf740  
14:27:46.389 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffd893c  A5      : 0x00000000  
14:27:46.389 -> A6      : 0x3ffd88f4  A7      : 0xffffffc8  A8      : 0x8008922e  A9      : 0x3ffdf700  
14:27:46.389 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000001  
14:27:46.436 -> A14     : 0x00060a20  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
14:27:46.436 -> EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
14:27:46.436 -> 
14:27:46.436 -> Backtrace: 0x40177b8f:0x3ffdf740 0x400ee04e:0x3ffdf760 0x400ee2a8:0x3ffdf780 0x40177abb:0x3ffdf7a0 0x400e58c9:0x3ffdf7c0 0x400e5931:0x3ffdf7f0 0x400e5a67:0x3ffdf810 0x40089479:0x3ffdf840

14:52:21.093 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 1329
14:52:21.602 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 13
14:52:21.602 ->   requested 40071
14:52:22.115 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 1330
14:52:22.678 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 13
14:52:23.101 ->   requested 499
14:52:23.616 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 1331
14:52:28.667 -> [W][AsyncTCP.cpp:962] _poll(): ack timeout 4
14:52:28.667 -> [V][esp32ModbusTCP.cpp:263] _onTimeout(): TCP timeout
14:52:28.667 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
14:52:28.667 -> Core 1 register dump:
14:52:28.667 -> PC      : 0x40177b8f  PS      : 0x00060430  A0      : 0x800ee051  A1      : 0x3ffdf7e0  
14:52:28.667 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffd89a4  A5      : 0x00000000  
14:52:28.667 -> A6      : 0x3ffd895c  A7      : 0xffffffc8  A8      : 0x8008922e  A9      : 0x3ffdf7a0  
14:52:28.667 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000001  
14:52:28.667 -> A14     : 0x00060420  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
14:52:28.667 -> EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
14:52:28.667 -> 
14:52:28.667 -> Backtrace: 0x40177b8f:0x3ffdf7e0 0x400ee04e:0x3ffdf800 0x400ee2a8:0x3ffdf820 0x40177abb:0x3ffdf840 0x400e58c9:0x3ffdf860 0x400e5931:0x3ffdf890 0x400e5a67:0x3ffdf8b0 0x40089479:0x3ffdf8e0

14:57:49.701 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 132
14:57:50.214 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 13
14:57:50.308 ->   requested 40071
14:57:50.688 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 133
14:57:55.231 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 13
14:57:55.327 ->   requested 499
14:57:55.703 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 134
14:57:56.036 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 21
14:57:56.322 ->   requested 40265
14:57:56.697 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 135
14:57:57.266 -> [V][esp32ModbusTCP.cpp:270] _onData(): data: len 17
14:57:57.314 ->   requested 40282
14:57:57.693 -> [V][esp32ModbusTCP.cpp:171] _tryToSend(): send id 136
14:58:02.733 -> [W][AsyncTCP.cpp:962] _poll(): ack timeout 4
14:58:02.733 -> [V][esp32ModbusTCP.cpp:263] _onTimeout(): TCP timeout
14:58:02.733 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
14:58:02.733 -> Core 1 register dump:
14:58:02.733 -> PC      : 0x40177b8f  PS      : 0x00060830  A0      : 0x800ee051  A1      : 0x3ffdf670  
14:58:02.733 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffd906c  A5      : 0x00000000  
14:58:02.733 -> A6      : 0x3ffd9024  A7      : 0xffffffc8  A8      : 0x8008922e  A9      : 0x3ffdf630  
14:58:02.733 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000001  
14:58:02.780 -> A14     : 0x00060820  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
14:58:02.780 -> EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
14:58:02.780 -> 
14:58:02.780 -> Backtrace: 0x40177b8f:0x3ffdf670 0x400ee04e:0x3ffdf690 0x400ee2a8:0x3ffdf6b0 0x40177abb:0x3ffdf6d0 0x400e58c9:0x3ffdf6f0 0x400e5931:0x3ffdf720 0x400e5a67:0x3ffdf740 0x40089479:0x3ffdf770

````

and the exception decoder output:

````
PC: 0x40177b8f: ModbusRequest::getId() const at lib\esp32ModbusTCP\src\esp32ModbusMessage.cpp line 92
EXCVADDR: 0x00000010

Decoding stack results
0x40177b8f: ModbusRequest::getId() const at lib\esp32ModbusTCP\src\esp32ModbusMessage.cpp line 92
0x400ee04e: esp32ModbusTCP::_clearQueue(esp32Modbus::Error) at lib\esp32ModbusTCP\src\esp32ModbusTCP.cpp line 197
0x400ee2a8: esp32ModbusTCP::_onTimeout(void*, AsyncClient*, unsigned int) at lib\esp32ModbusTCP\src\esp32ModbusTCP.cpp line 265
0x40177abb: std::_Function_handler ::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, unsigned int&&) at c:\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x400e58c9: AsyncClient::_poll(tcp_pcb*) at c:\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400e5931: AsyncClient::_s_poll(void*, tcp_pcb*) at lib\AsyncTCP\src\AsyncTCP.cpp line 1206
0x400e5a67: _async_service_task(void*) at lib\AsyncTCP\src\AsyncTCP.cpp line 171
0x40089479: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

´´´´

Can you take a look?, thanks.

bertmelis, 2020-05-28:
Looking into it. Multiple solutions to this issue.

Could you comment out clearing the `_toSend` queue in `_clearQueue`?

pablozg, 2020-05-29:
Hi, I had to change the function like this to avoid the "loadprohibited" error, because after comment out the first "_toSend  while", the error still happening.

````
void esp32ModbusTCP::_clearQueue(esp32Modbus::Error error) {
  if (xSemaphoreTake(_semaphore, 1000) == pdTRUE) {
    while(!_toSend.empty()) {
      //_tryError(_toSend.front().request->getId(), error, _toSend.front().arg);
      _toSend.pop_front();
    }
    while(!_toReceive.empty()) {
      //_tryError(_toSend.front().request->getId(), error, _toReceive.front().arg);
      _toReceive.pop_front();
    }
    xSemaphoreGive(_semaphore);
  } else {
    log_e("couldn't obtain semaphore");
    return;
  }
}
````