bertmelis/esp32ModbusTCP #18: Disconnecting or closing a link

CarlSchneider, 2020-07-06:
Hi bertmelis

Thanks for your efforts with the original ESP32ModbusTCP library.

My setup is a Sunny Boy and Sunny Island. Using your original library (not v1) I am able to read the holdingregisters I need using an ESP32 by creating two Modbus server instances. In the development environment I need to tweak code and upload, which on occasion causes Modbus comms errors to occur. If I kill the ESP32 and wait 3 minutes I get the Modbus back up and running correctly without errors. The .onerror function is reporting the errors but the link rarely recovers on its own. I’m polling 4 registers on the SI and one register on the SB every 1s.

To regain a ModbusTCP connection to the SMA without errors I either need to timeout the link (brute force) or use a library function to close the link and reopen it after a short delay(preferable as it’s more elegant).

I have scanned the ESP32ModbusTCP and AsyncTCP libraries for relevant .close() or similar .disconnect() commands, but can’t find any that seem to work they way I would have thought they worked. For example I want to close and reinstate the link ESP32ModbusTCP sunnyisland(3,{IP},502) by issuing a sunnyisland.close() on an error and then reinstate the link with another ESP32ModbusTCP sunnyisland(3,{IP},502) after a short delay.

Did you implement a TCP link disconnection in your original library? If so could you please share the syntax?

Many thanks
Carl

bertmelis, 2020-07-07:
Look at (private) method _disconnect:

https://github.com/bertmelis/esp32ModbusTCP/blob/8621d989f6b2c5886353fe3884af0a6818d3f452/src/esp32ModbusTCP.cpp#L107

CarlSchneider, 2020-07-07:
Hi bertmelis

Thank you very much for the rapid response. My knowledge is a very thin veneer and I have extremely limited knowledge of libraries.

I had a look at

void esp32ModbusTCP::_disconnect(bool now) {
  log_v("disconnecting");
  _state = DISCONNECTING;
  _client.close(now);
}

in the esp32ModbusTCP.cpp.

Based on the the other functions within the library I would have expected

sunnyisland.disconnect(true);

to be the appropriate function call to disconnect and close the sunnyisland client.

I’m working in VSCode and intellisens is indicating that
class “esp32ModbusTCP” has no member “disconnect”

Clearly I’m missing something and I’m hoping you could again please point me in the right direction.

I fiddled and I mean I’m fiddling in the dark, I tried

sunnyisland.~esp32ModbusTCP.disconnect(true);

and intellisens seems to like it :)

Hopefully I’m on the right track but if not more guidance please.

Many thanks
Carl

bertmelis, 2020-07-07:
You need to do a few steps:

make the function public. It's private now. So move the function in the .h file from the private section the the public section.

Maybe rename the function: remove the underscore. But you'll also have to change the name in all places that call the function (in the cpp file).

Does this make sense to you?

CarlSchneider, 2020-07-07:
Hi bertmelis

Thanks for the even more rapid response.

I’ll give it a bash and see if I get it right.

In reading the original library more closely I notice that writing to registers is not implemented. I’ll try my luck with this too.

Or would you recommend I rather move to v1 and adjust my sketch accordingly? I would still need to change the header to make the .disconnect() public, but it appears that the register write has been implemented.

Many thanks
Carl

CarlSchneider, 2020-07-08:
Hi bertmelis 

I followed your guidance above, thank you.
“You need to do a few steps: make the function public. It's private now. So move the function in the .h file from the private section the the public section.”

Done. I commented it out in the private section and pasted it into public after readInputRegisters line. 
I removed the underscore as suggested.

LBut you'll also have to change the name in all places that call the function (in the cpp file).”
There were only two instances where _disconnect is called in the .cpp file and I deleted the underscore to match the .h.

The intellisens was seeing the public function .connect and the sketch compiles.

However no matter what I use it does not seem to disconnect the link, as I continue to get read errors. I tried 
sunnyisland.disconnect();
sunnyisland.disconnect(true);
sunnyisland.disconnect(now);
sunnyisland.disconnect(false);

I then resorted to commenting out the sunnyisland.disconnect() and just did not read the holdingregister for 180s. This did not reset the link either, even though #define MB_IDLE_DISCONNECT_TIME is set to 10000ms. I then powered the ESP32 off for 180s and then powered it up. Now the link is up and the holdingregister is being successfully read. This physical power cycle with the 3 minute delay (2 minutes is too short) does it every time so far.

Any guidance would be appreciated.

Many thanks
Carl

CarlSchneider, 2020-07-11:
Hi bertmelis

I have installed v1 library, adjusted for the load prohibited error by commenting out the two lines in esp32ModbusTCP.cpp, adjusted code to examples, compiled and uploaded fine.

However the esp32 is now rebooting after the first SI holdingregister read per below.

SI reading error
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1ee1  PS      : 0x00060430  A0      : 0x800e41d4  A1      : 0x3ffde7a0  
A2      : 0x3ffde7ec  A3      : 0x3ffde7e4  A4      : 0x3ffde7e2  A5      : 0x3ffde7e3  
A6      : 0x3ffde7e8  A7      : 0x3ffde7e6  A8      : 0x3ffdecf1  A9      : 0x00000000  
A10     : 0x00000008  A11     : 0x00000001  A12     : 0x00000005  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x3ffbc830  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1ee1:0x3ffde7a0 0x400e41d1:0x3ffde7c0 0x400e45d5:0x3ffde810 0x4016507d:0x3ffde840 0x400e4e29:0x3ffde860 0x400e4e6d:0x3ffde8a0 0x400e52d6:0x3ffde8c0 0x40089805:0x3ffde8f0

Rebooting...

Hopefully this helps you to identify the problem.

Cheers
Carl